OpenStickCommunity/GP2040-CE #1477: Title: Feature Request: In-Controller Profile Switching

namreh72-del, 2025-09-03:
I really appreciate how affordable and capable the GP2040-CE platform is—it’s made high-performance setups accessible for users like me. One feature I’d love to see added is the ability to switch between multiple profiles directly on the controller, without needing to connect to a PC or web config. This would be a huge help for players who rotate between different games or platforms and want quick access to custom layouts or macros. Thanks for all the work you’ve put into this project—it’s already a game-changer.

---
thank you

Pelsin, 2025-09-03:
I recommend reading through the documentation, profile switching can also be done with hotkeys: https://gp2040-ce.info/hotkeys/

namreh72-del, 2025-09-03:
Thanks...im so new to this so appreciate